aadhithya14/Open-Teach #1: Questions about VR Headset

Nimolty, 2024-03-14:
Hello! Thanks for making this great works open-source.

I have a question. Is VR Headset Meta Quest 2 suitable for this system? 

Thank you so much!

aadhithya14, 2024-03-14:
Yes this codebase works with all Meta Quest devices. It supports Meta Quest 2 as well.

krishpop, 2024-03-17:
Hi! thanks for opensourcing this codebase, it seems really useful! When I load the apk on the quest 3 for the first time, the menu page is static and doesn't seem to respond to any controller/hand inputs. Is there a specific order in which the server and app have to start, or is it possibly a versioning-related issue. Any debugging support/advice would be appreciated!

aadhithya14, 2024-03-17:
Hi Krishnan, Thanks for raising this issue. As soon as you start the APK, you have to click the menu icon and there will be an option to change IP. Ensure to add your IP address of the network your robot/simulation is connected to using the drop downs. As soon as the IP is entered, enter go back and click Stream.  Similarly on the codebase side [Link](https://github.com/aadhithya14/Open-Teach/blob/main/configs/network.yaml) add the IP address. You should be able to use the robot/simulation using your hand. Feel free to email at aadhithya.iyer@nyu.edu for any more queries.

krishpop, 2024-03-17:
Still no luck here, stuck on a white page: 
[image]

krishpop, 2024-03-17:
Thanks for the help! It looks like hand-tracking was disabled and there was no warning/error in place to notice this.

aadhithya14, 2024-03-17:
No problem. The APK expects the user to have hand tracking enabled since the app is fully controlled by hand. Feel free to contact me in case of any issues in the future.

steb6, 2024-04-19:
Hi @aadhithya14 , I am trying to use your system on a Quest 2 as well. I managed to install everything and I was trying to launch the Libero simulation, anyway I have some questions:
- in the vr appication I can see the black hands over mine, but when I click 'stream' the panel becomes green and everything freezes. this happens without launching the server, this is the behaviour if the vr can't connect to the server?
- in the code the keypoint port is 8088 and in the vr app is 8087, is there a reason for this or I need to change the port?
- in the config file, host_address is the ip of the machine running the server, and left_xarm_ip is the one of the vr?

Thank you for releasing your code! :)

aadhithya14, 2024-04-19:
1) The APK client and the server runs together. Yes, this is the behaviour when closed loop control between the client and server us not established. So once you start the APK, start the server teleop as well using

steb6, 2024-04-22:
Thank you for your answer! I am sorry, but I still need a bit of clarification.
When I launch ```python teleop.py robot=libero_sim sim_env=True``` on a machine, that machine is the server and it launches the simulation as well, right? The host ip is the ip of the Quest, since I see from the code that it publishes the simulation images on that address, right?

steb6, 2024-04-22:
I would like to add that I have installed the repository inside my WSL2 on a Windows 11 host. I am setting te host_address in network.yaml to 0.0.0.0 and in the VR I put the IP of the Windows 11 host. Anyway, it seems that I can't read the pose of the hands. I noticed that sometimes the panel in the VR app becomes blu, orange, red, green, etc.

aadhithya14, 2024-04-23:
Hi Thanks for using Openteach. The host ip is the IP of the machine. You need to enter the machine IP on the server side as well as inside the APK. We used Ubuntu, but it should straightforward in WSL2 as well. Set your host_address in the configs to be your machine IP as well as set the IP in the VR to be your machine IP.

closed 2024-04-27 by aadhithya14