aadhithya14/Open-Teach #4: Issues during reappearance

JaceZhang1020, 2024-03-29:
Dear aadhithya14,

Greetings!

I am a user from China who is currently attempting to replicate the Simulation content outlined in the "Robot Controller Installation Specific Information" from your project. I am using a Quest 3 device with an Ubuntu 20.04 server system.

During my attempt to run the command `python3 teleop.py robot=libero_sim sim_env=True`, I encountered several issues. The specific error messages are as follows:

```
(openteach) ➜  Open-Teach git:(main) ✗ python3 teleop.py robot=libero_sim sim_env=True
***************************************************************
     Starting keypoint position transform component
***************************************************************
***************************************************************
     Starting vr detector component
***************************************************************
Process Process-2:
Process Process-3:
Traceback (most recent call last):
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 92, in _call_target
    return _target_(*args, **kwargs)
  File "/home/ghost/workbench/Open-Teach/openteach/components/detector/oculusbimanual.py", line 18, in __init__
    self.raw_keypoint_right_socket = create_pull_socket(host, oculus_right_port)
  File "/home/ghost/workbench/Open-Teach/openteach/utils/network.py", line 20, in create_pull_socket
    socket.bind('tcp://{}:{}'.format(host, port))
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/zmq/sugar/socket.py", line 302, in bind
    super().bind(addr)
  File "zmq/backend/cython/socket.pyx", line 564, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Cannot assign requested address (addr='tcp://172.24.71.206:8087')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ghost/workbench/Open-Teach/openteach/components/initializers.py", line 112, in _start_component
    component = hydra.utils.instantiate(configs)
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 339, in instantiate_node
    return _call_target(_target_, partial, args, kwargs, full_key)
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 97, in _call_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error in call to target 'openteach.components.detector.oculusbimanual.OculusVRTwoHandDetector':
ZMQError('Cannot assign requested address (addr='tcp://172.24.71.206:8087')')
full_key: robot.detector
Traceback (most recent call last):
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 92, in _call_target
    return _target_(*args, **kwargs)
  File "/home/ghost/workbench/Open-Teach/openteach/components/detector/keypoint_transform.py", line 16, in __init__
    self.transformed_keypoint_publisher = ZMQKeypointPublisher(host, transformation_port)
  File "/home/ghost/workbench/Open-Teach/openteach/utils/network.py", line 39, in __init__
    self._init_publisher()
  File "/home/ghost/workbench/Open-Teach/openteach/utils/network.py", line 44, in _init_publisher
    self.socket.bind('tcp://{}:{}'.format(self._host, self._port))
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/zmq/sugar/socket.py", line 302, in bind
    super().bind(addr)
  File "zmq/backend/cython/socket.pyx", line 564, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Cannot assign requested address (addr='tcp://172.24.71.206:8089')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 108, in run
[... 165 lines cut ...]
  File "/home/ghost/workbench/Open-Teach/openteach/components/initializers.py", line 112, in _start_component
    component = hydra.utils.instantiate(configs)
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 325, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET), full_key)
  File "/home/ghost/miniconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 139, in _resolve_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error locating target 'openteach.components.operators.libero_sim.LiberoSimOperator', see chained exception above.
full_key: robot.operators0
(openteach) ➜  Open-Teach git:(main) ✗ 
```

Based on my understanding, these issues may be related to ZMQ address binding and EGL device display initialization. I have tried the suggestions from ChatGPT, but I have not been able to resolve the issues.

For the ZMQ address binding error, I would like to know how to correctly obtain the port numbers for the headset and the two controllers. Are there any specific tools or commands that can assist me in diagnosing and resolving this issue?

Regarding the EGL device display initialization error, I understand that it may be due to the absence of an EGL driver that supports the PLATFORM_DEVICE extension. Could you please advise me on how to install or update this driver so that I can successfully initialize the EGL device display?

I am eagerly awaiting your response and suggestions and hope to resolve these issues as soon as possible to continue with my project research.

Thank you very much for your time and assistance!

Best regards,
JaceZhang1020

aadhithya14, 2024-03-29:
Hi ,
Thanks for raising this issue. As mentioned in the [docs ](https://github.com/aadhithya14/Open-Teach/blob/main/docs/teleop_data_collect.md), you have to enter your IP of the network. Right now the IP is set at 172.24.71.206. Change this to your machine IP. That should solve the problem of ports. For the problem of EGL drivers, try running the teleoperation using MUJOCO_GL=egl python teleop.py sim_env=True.

JaceZhang1020, 2024-04-03:
Hi,
Thank you for your reply.I wonder if I need to change the IP address of hand controllers as well please,since I also saw them in ```network.yaml```.Meanwhile,I wanna make sure if you mean that I should use command-line parameter ```MUJOCO_GL=egl sim_env=True``` while running python teleop.py?

aadhithya14, 2024-04-04:
Hi,
You don't need to change anything except the IP of the network if you are just running the simulation. And to answer the other question yes, try `MUJOCO_GL=egl python teleop.py sim_env=True.` and let me know if that works. 
Best,
Aadhithya Iyer

JaceZhang1020, 2024-04-06:
Understood.I will try it,thank you again~

aadhithya14, 2024-04-18:
Hi ,
I wanted to check if you were able to get things running on Libero environment. Looking forward to hearing from you. 

Best,
Aadhithya Iyer

JaceZhang1020, 2024-04-19:
Hi，

Thank you for thinking about this thing all the time.Recently I am busy on my exams so have no time on teleoperation for the moment,so I make no process.If I meet new problem about previous,I will give you feedback ASAP.

Thank you again for asking :)

Best,
JaceZhang1020

获取Outlook for Android<https://aka.ms/AAb9ysg>

closed 2024-06-03 by aadhithya14