aadhithya14/Open-Teach #6: error when running python3 teleop.py robot=liberos_sim sim_env=True

ChangerC77, 2024-04-26:
Here is my error output:
```
(openteach) ➜  Open-Teach git:(main) ✗ python teleop.py robot=libero_sim sim_env=True
***************************************************************
     Starting vr detector component
***************************************************************
***************************************************************
     Starting keypoint position transform component
***************************************************************
Process Process-1:
Traceback (most recent call last):
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/utils.py", line 639, in _locate
    obj = getattr(obj, part)
AttributeError: module 'openteach.components.environment' has no attribute 'libero_env'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/utils.py", line 645, in _locate
    obj = import_module(mod)
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/ros/GAIRLAB/Open-Teach/openteach/components/environment/libero_env.py", line 11, in <module>
    from libero.libero import benchmark, get_libero_path
ModuleNotFoundError: No module named 'libero'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 134, in _resolve_target
    target = _locate(target)
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/utils.py", line 648, in _locate
    raise ImportError(
ImportError: Error loading 'openteach.components.environment.libero_env.LiberoEnv':
ModuleNotFoundError("No module named 'libero'")
Are you sure that 'libero_env' is importable from module 'openteach.components.environment'?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ros/GAIRLAB/Open-Teach/openteach/components/initializers.py", line 112, in _start_component
    component = hydra.utils.instantiate(configs)
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 325, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET), full_key)
  File "/home/ros/anaconda3/envs/openteach/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 139, in _resolve_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error locating target 'openteach.components.environment.libero_env.LiberoEnv', see chained exception above.
full_key: robot.environment0
***************************************************************
     Starting hand 2D plotter component
***************************************************************
***************************************************************
     Starting libero operator component
***************************************************************
***************************************************************
     Starting calibration component
***************************************************************

Calibration file already exists. Do you want to create a new one? Press y for Yes else press Entern
Stopping the calibrator.
Closing the subscriber socket in [192.168.2.110:8089](http://192.168.2.110:8089/).
THUMB BOUNDS IN THE OPERATOR: [array([0.07, 0.1 ]), array([0.07, 0.07]), array([-0.01,  0.08]), array([-0.01,  0.09]), [-0.029687233564084578, -0.060893850578688846]]
***************************************************************
     Starting Libero_Sim control component
***************************************************************
Start controlling the robot hand using the Oculus Headset.

****** RESETTING TELEOP ****** 
/home/ros/GAIRLAB/Open-Teach/openteach/components/visualizers/plotters/plotter_2d.py:91: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.pause(0.001) # This make the graph show up if matplotlib is in Tkinter mode
```
and I use BimanualArm.apk, but now it can only recognize my right hand pose, does it ok? 
and then I run python3 data_collect.py robot=libero_sim sim_env=True demo_num=1, it will output this inofrmation:
```
Reaching correct function
***************************************************************
     Starting RGB stream: 10005 component
***************************************************************
tcp://[192.168.2.110:10005](http://192.168.2.110:10005/)
***************************************************************
     Starting Depth stream: 11005 component
***************************************************************
tcp://[192.168.2.110:11005](http://192.168.2.110:11005/)
Starting to record RGB frames from port: 10005
***************************************************************
     Starting endeffector_cartesian_states component
***************************************************************
Checking if the keypoint port is active...
Starting to record keypoints to store in extracted_data/demonstration_1/endeffector_cartesian_states.h5.
***************************************************************
     Starting endeffector_commanded_cartesian_states component
***************************************************************
Checking if the keypoint port is active...
Starting to record keypoints to store in extracted_data/demonstration_1/endeffector_commanded_cartesian_states.h5.
```
but I cannot see any simulation start, did I make a step mistake somewhere?

ChangerC77, 2024-04-26:
I found that it occurred this error: ```from libero.libero import benchmark, get_libero_path
ModuleNotFoundError: No module named 'libero'``` but I downloaded LIBERO library in open-tech file, Did I put it into the wrong path?

ChangerC77, 2024-04-29:
ok, I have solved these problems. There are several places that need to be modified. One is the path of libero. The one cloned from github is LIBERO/libero. Then I only need to put libero into Open tech, and in /libero/libero/__init__ You need to modify the path of libero in .py, and then use from openteach.components.environment.libero_env import LiberoEnv to create the dataset path in python.

closed 2024-04-29 by ChangerC77

aadhithya14, 2024-04-29:
Hi yes, you are right and sorry for missing this issue. This information has been added in the docs to enter the dataset path in python.

boreep, 2024-11-28:
> and I use BimanualArm.apk, but now it can only recognize my right hand pose, does it ok?
> but I cannot see any simulation start, did I make a step mistake somewhere?

Hi,I have the same issue, here's the output after running `teleop.py`

```
(openteach) (base) hxd@5600g:~/Open-Teach-main$ python3 teleop.py robot=libero_sim sim_env=True
***************************************************************
     Starting vr detector component
***************************************************************
***************************************************************
     Starting keypoint position transform component
***************************************************************
***************************************************************
     Starting hand 2D plotter component
***************************************************************
***************************************************************
     Starting libero operator component
***************************************************************
***************************************************************
     Starting calibration component
***************************************************************

Calibration file already exists. Do you want to create a new one? Press y for Yes else press Entertcp://192.168.0.153:10005
tcp://192.168.0.153:10006
tcp://192.168.0.153:11005
tcp://192.168.0.153:11006
Initializing Environment
[Warning]: datasets path /home/hxd/Open-Teach-main/LIBERO/libero/libero/../datasets does not exist!
***************************************************************
     Starting Libero_Sim environment component
***************************************************************
/home/hxd/Open-Teach-main/openteach/components/visualizers/plotters/plotter_2d.py:91: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.pause(0.001) # This make the graph show up if matplotlib is in Tkinter mode
```
but I still cannot see any simulation start,I would like to know more details about your solution to this issue。
[image]