aadhithya14/Open-Teach #9: Arm compatibility

arthurnni, 2024-04-30:
Can I use xArm 5   5DOF  with  your  project ?

What  model  of  RealSense  and Fisheye cameras used  in your  project ?

Where  to place  the cameras ?

aadhithya14, 2024-05-01:
Yes, you can use xArm 5 but you might need to write out some wrappers for the robot and operator to have the retargeting done. We provide templates for how to write the retargeting code and robot wrappers.

arthurnni, 2024-05-01:
What exact model  of xArm , RealSense and FishEye camera are  you  using  in  your  project ?

aadhithya14, 2024-05-01:
We use 6DOF Xarm , Realsense D435i and A simple USB camera as the fisheye camera

arthurnni, 2024-05-02:
How many  D435i and FishEye cameras  are used  in your  project ?   Where  are those cameras  located/installed ?
There is  no picture  show  those  cameras  in your  websites.

aadhithya14, 2024-05-13:
We used 4 D435i and 2 Fisheye in our setups at max for the project. But you can add as many cameras as possible if your internet is decent enough. We have now setups scaling to upto 6 - 8 cameras.

closed 2024-07-03 by aadhithya14