aadhithya14/Open-Teach #10: git LFS issue when cloning

blooop, 2024-05-01:
I am getting git-lfs smudge errors when I try to clone.  Does the error message look like a file you would expect to be cloned?

```
git clone https://github.com/aadhithya14/Open-Teach.git
Cloning into 'Open-Teach'...
remote: Enumerating objects: 6951, done.
remote: Counting objects: 100% (143/143), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 6951 (delta 87), reused 126 (delta 78), pack-reused 6808
Receiving objects: 100% (6951/6951), 511.87 MiB | 9.54 MiB/s, done.
Resolving deltas: 100% (3617/3617), done.
Updating files: 100% (12041/12041), done.
Downloading VR/Bimanual-Robot-Unity/Assets/Oculus/SampleFramework/Usage/Firebase/README.md (1.3 KB)
Error downloading object: VR/Bimanual-Robot-Unity/Assets/Oculus/SampleFramework/Usage/Firebase/README.md (28b6429): Smudge error: Error downloading VR/Bimanual-Robot-Unity/Assets/Oculus/SampleFramework/Usage/Firebase/README.md (28b6429da3cdbba903ab35ca4bdc6d3621ffcb8224a36564d17593bcf8b3ab9d): [28b6429da3cdbba903ab35ca4bdc6d3621ffcb8224a36564d17593bcf8b3ab9d] Object does not exist on the server: [404] Object does not exist on the server

Errors logged to /home/xxxxxx/projects/Open-Teach/.git/lfs/logs/20240501T123943.0248752.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: VR/Bimanual-Robot-Unity/Assets/Oculus/SampleFramework/Usage/Firebase/README.md: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

my git-lfs version.  

```
git-lfs -v
git-lfs/3.0.2 (GitHub; linux amd64; go 1.18.1)
```

blooop, 2024-05-03:
Thanks.  That is an improvement as the repo is now checked out and in a clean state, but i think the underlying issue is still there.

```
git clone https://github.com/aadhithya14/Open-Teach.git
Cloning into 'Open-Teach'...
remote: Enumerating objects: 6951, done.
remote: Counting objects: 100% (143/143), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 6951 (delta 87), reused 126 (delta 78), pack-reused 6808
Receiving objects: 100% (6951/6951), 511.87 MiB | 19.30 MiB/s, done.
Resolving deltas: 100% (3617/3617), done.
Updating files: 100% (12041/12041), done.
Encountered 2 file(s) that should have been pointers, but weren't:
	VR/Bimanual-Robot-Unity/README.md
	VR/Franka-Bot-Unity/README.md

```
But when I try to pull the lfs files they still can't be found
```
cd Open-Teach/; git lfs pull
[ffb934fc9f094ef320fc8d38eb141d8d962abdf92c8d4a2175011f94ac59fa18] Object does not exist on the server: [404] Object does not exist on the server
[033c03a7178a6763f9a8c644f8c2c8f7ddb167cd53d084ee1ff655b2d1b92634] Object does not exist on the server: [404] Object does not exist on the server
[e6c0777c2c08befdb34ebd130ef874a4a4d086667b658d782c1b218316b7ecf3] Object does not exist on the server: [404] Object does not exist on the server
[28b6429da3cdbba903ab35ca4bdc6d3621ffcb8224a36564d17593bcf8b3ab9d] Object does not exist on the server: [404] Object does not exist on the server
error: failed to fetch some objects from 'https://github.com/aadhithya14/Open-Teach.git/info/lfs'

```

aadhithya14, 2024-05-10:
Yes you can take those readme's out. Those readme's come together with the unity files. I will stash those. Apart from that you should be able to use the APK without the need of the unity codefiles. The unity codefiles are needed in case you want to make any changes to the source code of APK.

closed 2024-07-03 by aadhithya14