aadhithya14/Open-Teach #11: Issues in simulation

Norweig1an, 2024-05-08:
Hi, thanks for your inspiring work!
During simulation, we encountered some issues:
The IP and ports were correctly set.
When we run `python teleop.py robot=allegro_sim sim_env=True` the scene in Meta Quest 3 froze after the hand model was loaded.
It seems that the server `joint_angle_subscriber` could not receive `joint_angles` (openteach/components/environment/allegro_hand_env.py, `take_action(self)').
We will appreciate your kind help.

chiawenwang, 2024-05-09:
I also meet the same problem on Ubuntu20.04 !

aadhithya14, 2024-05-09:
Thanks for using Openteach. Hi Could you print the screenshot of your terminal? Is the end effector position getting printed?

chiawenwang, 2024-05-10:
> Thanks for using Openteach. Hi Could you print the screenshot of your terminal? Is the end effector position getting printed?

Thank you for you replying. It just happens when I use the **Bimanual**, and it works on  KinovaBot. Here is the content printed:

******************
(openteach_isaac) wangjw@TRX40-AORUS-PRO:~/Open-Teach$ python3 teleop.py robot=allegro_sim sim_env=True
***************************************************************
     Starting keypoint position transform component
***************************************************************
***************************************************************
     Starting vr detector component
***************************************************************
Importing module 'gym_37' (/home/wangjw/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/wangjw/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
Importing module 'gym_37' (/home/wangjw/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/wangjw/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
***************************************************************
     Starting hand 2D plotter component
***************************************************************
PyTorch version 1.13.1+cu117
Device count 1
/home/wangjw/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/wangjw/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Emitting ninja build file /home/wangjw/.cache/torch_extensions/py37_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
tcp://192.168.31.176:10005
tcp://192.168.31.176:11005
Trying to make simulation
PyTorch version 1.13.1+cu117
Device count 1
/home/wangjw/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/wangjw/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Not connected to PVD
+++ Using GPU PhysX
Emitting ninja build file /home/wangjw/.cache/torch_extensions/py37_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
Loading extension module gymtorch...
***************************************************************
     Starting allegro hand sim operator component
***************************************************************
/home/wangjw/.conda/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link Base link (index: 0) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/wangjw/.conda/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_3.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/wangjw/.conda/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_7.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/wangjw/.conda/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_11.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/wangjw/.conda/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_15.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
***************************************************************
     Starting calibration component
***************************************************************
Calibration file already exists. Do you want to create a new one? Press y for Yes else press EnterSimulation created
[... cut ...]
THUMB BOUNDS IN THE OPERATOR: [array([0.07147029, 0.10102608]), array([0.06931292, 0.06902901]), array([-0.00888068,  0.08024863]), array([-0.0074921,  0.0941279]), [-0.029687233564084578, -0.060893850578688846]]
***************************************************************
     Starting Allegro_Sim control component
***************************************************************
Start controlling the robot hand using the Oculus Headset.
Applying retargeted angles
******************

The image shows the hands are frozon after `python3 teleop.py robot=allegro_sim sim_env=True`
[image]

Norweig1an, 2024-05-10:
> Thanks for using Openteach. Hi Could you print the screenshot of your terminal? Is the end effector position getting printed?

python teleop.py robot=allegro_sim sim_env=True
Importing module 'gym_37' (/home/rhos/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rhos/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
/home/rhos/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/torch/utils/cpp_extension.py:3: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
PyTorch version 1.7.1+cu110
Device count 1
/home/rhos/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/rhos/.cache/torch_extensions as PyTorch extensions root...
Emitting ninja build file /home/rhos/.cache/torch_extensions/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
tcp://192.168.0.188:10005
tcp://192.168.0.188:11005
Trying to make simulation 0 0 0 True 11.0 10de:2786
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
Simulation created
Loading asset 'allegro_hand_description/urdf/model_only_hand.urdf' from '/home/rhos/Desktop/Open-Teach/openteach/components/environment/assets/urdf/'
Loading Assets
Env suite is not None
***************************************************************
     Starting Allegro_Sim environment component
***************************************************************
Start controlling the Simulation hand using the Oculus Headset.

End Effector Position [ 1.         1.2        0.        -0.7071068 -0.7071068  0.
  0.       ]

Then the terminal stops printing.

aadhithya14, 2024-05-10:
Hi @chiawenwang , the allegro sim can be only used with the KinovaBot because we have only one robot hand in the simulation. We use the Bimanual APK for Bimanual gripper and Libero Simulation. So if you are able to use the allegro simulation with Kinova Bot you are all set and everything is perfect.

chiawenwang, 2024-05-11:
> Hi @chiawenwang, the allegro simulation can only be used with KinovaBot as we only have one robot in the simulation. We are using Bimanual APK for Bimanual robot and Libero simulation. So if you can use allegro simulation with Kinova Bot, then everything is ready and everything is perfect.

Thank you for your detailed reply! We are planning to use our own robot in LIBERO, do you have some instructions?

aadhithya14, 2024-05-13:
In the config you have the option to change the suite, [Link](https://github.com/aadhithya14/Open-Teach/blob/main/configs/robot/libero_sim.yaml) you can choosever whichever suite of tasks you want in libero. Also if you plan to add your own robot, which follows the axis coordinate system of libero you can directly use the current codebase for your robot.

chiawenwang, 2024-05-14:
> In the config you have the option to change the suite, [Link](https://github.com/aadhithya14/Open-Teach/blob/main/configs/robot/libero_sim.yaml) you can choosever whichever suite of tasks you want [...]

OK! Thank you so much! There is no problem for me now!

Norweig1an, 2024-05-17:
HI @aadhithya14 , I apologize for the delayed response. We tested on different PCs and received the same outcome: we are still unable to transfer data from VR Quest to the server. We identify the issue in 

> line 171 VR/Franka-Bot-Unity/Assets/Scripts/Gesture Detection/GestureDetector.cs
`client.sendFrame(RightHandDatastring);`
and 
> line 75 openteach/utils/network.py
`raw data =self.socket.recv()`

Although it appears that the data has been successfully sent by VR, none of it can be received by the PC. We've tried reinstall the APK and clean the data, but neither of them worked. Do you have any idea?

aadhithya14, 2024-05-21:
Are both quest and the machine under the same network provider. This is necessary because the APK expects it in that format.

Norweig1an, 2024-05-21:
Thanks for your reply! We're sure we have set all devices under the same network. And now we have discovered that we can transfer 2D plotter but hand angel. We still don't know the reason.

chenshuo20, 2024-06-15:
Hi, thank you for your help! @aadhithya14 

I am experiencing the same issue as @Norweig1an. I tried reinstalling the APK and ensured that both the Quest and the machine are on the same network, but it still doesn't work. My server gets stuck after printing "End Effector Position [1. 1.2 0. -0.7071068 -0.7071068 0. 0.]" and the Quest also becomes unresponsive.

Do you have any advice on how to resolve this issue? Thank you!

kopiotrek, 2024-09-06:
Hi, I am having the same issue!

aadhithya14, 2024-10-05:
This could be because of network instabilities. Try clearing the cache of the apk and start it. Or just reinstall the APK.

Bailey-24, 2024-10-13:
Hello,
I have some problem running in the simulation(I didn't have any real robot).
I wonder why the issac gym not show up and can't collect the data.

I enter the VR and put my own computer in to the IP address (because i have no real robot) and enter "Stream".
[image]

It works well. But the issac gym not show up. How to solve.
Here are the output in the terimintor.
```
(openteach_isaac) (base) pi@pi:~/Documents/Open-Teach$ python3 teleop.py robot=allegro_sim sim_env=True
***************************************************************
     Starting keypoint position transform component
***************************************************************
***************************************************************
     Starting vr detector component
***************************************************************
Importing module 'gym_37' (/home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
Importing module 'gym_37' (/home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
***************************************************************
     Starting hand 2D plotter component
***************************************************************
PyTorch version 1.13.1+cu117
PyTorch version 1.13.1+cu117
Device count 1
/home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/pi/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Device count 1
/home/pi/Documents/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/pi/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Emitting ninja build file /home/pi/.cache/torch_extensions/py37_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
tcp://192.168.1.4:10005
tcp://192.168.1.4:11005
Trying to make simulation
Loading extension module gymtorch...
***************************************************************
     Starting allegro hand sim operator component
***************************************************************
Not connected to PVD
+++ Using GPU PhysX
/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link Base link (index: 0) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_3.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_7.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_11.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_15.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
***************************************************************
     Starting calibration component
***************************************************************

Calibration file already exists. Do you want to create a new one? Press y for Yes else press Enter
Stopping the calibrator.
[... cut ...]
Stored the metadata in extracted_data/demonstration_1/cam_0_rgb_video.metadata.
    return self.poll(os.WNOHANG if timeout == 0.0 else 0)
  File "/home/pi/anaconda3/envs/openteach_isaac/lib/python3.7/multiprocessing/popen_fork.py", line 28, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt
Saved keypoint data in extracted_data/demonstration_1/allegro_commanded_joint_states.h5.
Saved keypoint data in extracted_data/demonstration_1/allegro_joint_states.h5.
```

How to solve these two problem. Thank you.

Bailey-24, 2024-10-19:
I reinstall dirver and then solve the problem 1

sernostine, 2024-12-03:
May I ask what driver you reinstalled? i kinda met the same problem,i have the skeletons on left top corner but the Sim window  doesn't show up. Should this Isaac Gym simulation window appear on the server/PC side or in VR?

Mayunsheng-1, 2025-01-13:
Reaching correct function
***************************************************************
     Starting RGB stream: 10005 component
***************************************************************
tcp://192.168.0.212:10005
***************************************************************
     Starting Depth stream: 11005 component
***************************************************************
tcp://192.168.0.212:11005
Starting to record RGB frames from port: 10005
***************************************************************
     Starting endeffector_cartesian_states component
***************************************************************
Checking if the keypoint port is active...
Starting to record keypoints to store in extracted_data/demonstration_1/endeffector_cartesian_states.h5.
***************************************************************
     Starting endeffector_commanded_cartesian_states component
***************************************************************
Checking if the keypoint port is active...
Starting to record keypoints to store in extracted_data/demonstration_1/endeffector_commanded_cartesian_states.h5.

May I ask why the APK software does not respond and the screen does not output graphics after connecting to Quest 3 after my terminal outputs this?

jingyi-xiang, 2025-06-29:
Hi, @Norweig1an and @chenshuo20 did you guys ever find a solution? I am running into the same issue where the PC never receives the data, and the last thing printed in the terminal is the end effector position. The code seems to get stuck on [line 497](https://github.com/aadhithya14/Open-Teach/blob/main/openteach/components/environment/allegro_hand_moving_env.py#L497) of `allegro_hand_moving_env.py`.

xiaoen0, 2025-07-11:
> Hi, [@Norweig1an](https://github.com/Norweig1an) and [@chenshuo20](https://github.com/chenshuo20) did you guys ever find a solution? I am running into the same issue where the PC never receives the [...]

@jingyi-xiang Were you able to resolve the issue where it got stuck?

jingyi-xiang, 2025-07-11:
I was able to get it to run after making a bunch of modifications. My understanding of the code (which could be wrong) is that the operator / retargeting process and the simulator process run in parallel, and they pass information to each other through socket connections. The only things passed between the headset and the pc are tracked hand poses and keypoints, and they seem to work fine during my testing. My current solution is to get rid of the operator class completely and integrate it into the simulator class to perform retargeting. I'm working from a private fork but I uploaded my modified code here: https://drive.google.com/file/d/1z7dKallG943Vrl21YwyR-4w4D1JvmyJk/view?usp=sharing