aadhithya14/Open-Teach #12: Starting the camera sensors

chiawenwang, 2024-05-14:
When I follow the [instruction](https://github.com/aadhithya14/Open-Teach/blob/main/docs/sensors.md) to stream the realsense into the VR, meeting the error in the camera_log:

Process Process-1:
Traceback (most recent call last):
  File "/home/wangjw/.conda/envs/openteach_wangjw/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/wangjw/.conda/envs/openteach_wangjw/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wangjw/Open-Teach/openteach/components/initializers.py", line 37, in _start_component
    component = RealsenseCamera(
  File "/home/wangjw/Open-Teach/openteach/components/sensors/realsense.py", line 39, in __init__
    self._start_realsense(self._cam_serial_num)
  File "/home/wangjw/Open-Teach/openteach/components/sensors/realsense.py", line 63, in _start_realsense
    cfg = self.pipeline.start(config)
RuntimeError: Couldn't resolve requests
tcp://192.168.31.176:10005
tcp://192.168.31.176:11005

Here is my camera.yaml:

```
defaults:
  - _self_
  - network
  - override hydra/hydra_logging: disabled  
  - override hydra/job_logging: disabled 

robot_cam_serial_numbers:
  - '243322074047'

cam_configs:
  width: 1280
  height: 720
  fps: 30
  processing_preset: 1 # High accuracy mode
  rotation_angle: 0

oculus_cam: 0 # First camera
num_cams: 1

hydra:  
  output_subdir: null  
  run:  
    dir: .
```

chiawenwang, 2024-05-15:
Fine, I should use USB3 to connect the realsense!

chiawenwang, 2024-05-15:
As to the fisheye camera, what is the **fisheye_cam_numbers**?
```
defaults:
  - _self_
  - network
  - override hydra/hydra_logging: disabled  
  - override hydra/job_logging: disabled 

fisheye_cam_numbers:
  - 7

cam_configs:
  width: 640
  height: 480
  fps: 30
  

oculus_cam: 0 # First camera
stream_oculus: False

hydra:  
  output_subdir: null  
  run:  
    dir: .
```

I try to find the right number by
```
    def _start_fisheye(self):
        
        for i in range(100000000000):
            self.cam_id = i    
            print("Cam Id is ", self.cam_id)
            self.cap = cv2.VideoCapture(self.cam_id)
            if self.cap.isOpened():
                print("find it! Cam Id is:", self.cam_id)  
                break  
            else:
                # self.cap.release() 
                print("trying Cam Id", self.cam_id, "Failed") 
```
But it failed.

aadhithya14, 2024-05-18:
Fisheye came numbers are the USB ports of the fisheye camera connected

closed 2024-06-05 by chiawenwang