aadhithya14/Open-Teach #14: Can't control  franka hand gripper(franka emika)

loli11111q, 2024-05-31:
i have already built the openteach project and successfully used meta quest2 to control my Franka arm.Although i am not very familiar with gesture control at the moment,i am confident that i can indeed control Franka arm through VR.But i can't control my FRANKA hand gripper,no matter what gesture i use,do i need to change any code? How can i control the gripper. At the same time,i hope to receive some suggestions on gesture operation,such as how to better control the robotic arm  with hands. 
  I am very grateful for any help.

aadhithya14, 2024-06-03:
Thanks for using Openteach. Sorry for the late reply. In our setup we had franka with Allegro hence the gripper control gesture has not been included but including the gripper is very easy and straightforward in our setup. Check out the function here in bimanual wrapper which has gripper control [here](https://github.com/aadhithya14/Open-Teach/blob/main/openteach/components/operators/bimanual_right.py#L243). Here we use pinch between pinky and thumb on the right hand as the gripper closing and opening.(But you can choose to have whatever you want). Make sure to take the corresponding class variables in this function from the init. Let me know if you have any other issues. Happy to help.

loli11111q, 2024-06-04:
Thank you for your reply. I can now control the gripper. At present, I have two more questions.

 Firstly, I would like to ask if you have any skills in controlling a single arm. Although I can control it now, I am very inexperienced and it is difficult for me to complete some simple tasks using my hands to control the robotic arm.

Secondly, I am unable to transmit my camera (Realsense) scene to the whiteboard in the headquest2 apk. I have successfully run the command bash launch_server.sh about opening the camera, and there have been no errors. I have also changed the relevant parameters. I can see the camera shooting scene on 0.0.0.0:5000, but I cannot push it to the whiteboard.

I would greatly appreciate any help.

aadhithya14, 2024-10-05:
Check if the VR is connected to internet. The IP should be the IP of the machine to which robot is connected. And if both of these things work check the image resolution. If the resolution is too high it might fail. But it works till 1280x720