aadhithya14/Open-Teach #17: can't collect data

loli11111q, 2024-06-17:
My  teleoperation is completely normal. Below are some screenshots of me and I hope you can give some suggestions.
   1.The following two images are screenshots of starting the arm and gripper servers on Nuc, respectively.
[image]
 
[image]
      2.The following image is a screenshot of me running collect_data. py
[image]
[image]
[image]

chiawenwang, 2024-07-25:
请问你现在可以收集数据了吗？

max-yhy, 2025-06-23:
> My teleoperation is completely normal. Below are some screenshots of me and I hope you can give some suggestions. 1.The following two images are screenshots of starting the arm and gripper servers [...]

[image]
why do i got this