aadhithya14/Open-Teach #18: Question about _get_hand_dir_frame method

xbkaishui, 2024-06-21:
Hello： 

Thanks for you great work, I have some questions about the detector logic method of  _get_hand_dir_frame

>    # Create a coordinate frame for the arm 
    def _get_hand_dir_frame(self, origin_coord, index_knuckle_coord, pinky_knuckle_coord):

        palm_normal = normalize_vector(np.cross(index_knuckle_coord, pinky_knuckle_coord))   # Unity space - Y
        palm_direction = normalize_vector(index_knuckle_coord + pinky_knuckle_coord)         # Unity space - Z
        cross_product = normalize_vector(index_knuckle_coord - pinky_knuckle_coord)              # Unity space - X
        
        return [origin_coord, cross_product, palm_normal, palm_direction]

I saw the  cross_product is x-axis,  why the x-axis is not cross y and z axis ?

aadhithya14, 2024-07-02:
you can do it either ways. If you think about we want to get the thumb direction, which can be estimated using cross of normal and palm direction as well as using subtraction. Here subtraction was used because unity operates in left hand frame and robots operate in right hand frame and to avoid frame mismatch issues we use subtraction rather than cross product. But even if you use cross product the result should be the same

xbkaishui, 2024-07-10:
ok got it thanks

closed 2024-07-10 by xbkaishui