aadhithya14/Open-Teach #19: How to control the arm and gripper in simulation

FinnJob, 2024-06-26:
Thank you for your excellent work! I have successfully launched the server using the command python teleop.py robot=libero_sim sim_env=True, and the BimanualArm.apk has successfully displayed the Libero environment and the hand gestures drawn in the upper left corner. However, no matter how I move, the robotic arm in Libero does not respond; only the gestures in the upper left corner change accordingly. Could you please guide me on how to effectively control the robotic arm and gripper? Additionally, if I want to control my own real robotic arm using OpenTeach, how should I configure it? Is the control method consistent with the simulation?

Thank you very much for your time and assistance!

FinnJob, 2024-06-27:
Hi, after a brief review of the code, it seems that OpenTeach cannot obtain head data for control, right?

aadhithya14, 2024-07-02:
Sorry for the late reply. Openteach cannot obtain head data for control. We are working on it right now, you might soon see it in future. Were you able to solve libero control issue?

ZuWang95, 2024-07-02:
> Thank you for your excellent work! I have successfully launched the server using the command python teleop.py robot=libero_sim sim_env=True, and the BimanualArm.apk has successfully displayed the [...]

Make sure to pinch your left thumb and middle finger till the screen is blue then try to start moving your arm and hand. For better control, match the initial pose of the hand to that in the simulation.

FinnJob, 2024-07-03:
@aadhithya14 Thank you for your response. Yes, I have been able to control the robotic arm in Libero, but it seems difficult to control the robotic arm to complete the grasping task. I have tried many times, but have failed.

aadhithya14, 2024-07-03:
For libero the control is you imagine you are holding the end effector of the robot and not mimicking your hand actions.

FinnJob, 2024-07-04:
@aadhithya14 OK, thanks, I will try it.

closed 2024-07-04 by FinnJob