aadhithya14/Open-Teach #20: [Error] [carb.gym.plugin] Failed to resolve visual mesh '../meshes/table/robot_stand.obj'

ligerfotis, 2024-07-17:
I get the following error when running python3 teleop.py robot=allegro_sim sim_env=True. Any idea what may cause it? The directory /meshes/table/robot_stand.obj does not exist in the repo.
The procedure I followed to install Isaac gym was:
- to create an environment with `conda env create -f env_isaac.yml` and activate it
- download Isaac gym preview release from [here](https://developer.nvidia.com/isaac-gym)
- navigate to the python folder in it and install it via `pip install -e .` 
- then exported everything needed, as described [here](https://github.com/aadhithya14/Open-Teach/blob/main/docs/simulation.md)
- finally run `python3 teleop.py robot=allegro_sim sim_env=True`

### Error Message
python3 teleop.py robot=allegro_sim sim_env=True
\***************************************************************
     Starting vr detector component
\***************************************************************
\***************************************************************
     Starting keypoint position transform component
\***************************************************************
Importing module 'gym_37' (/home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
Importing module 'gym_37' (/home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
\***************************************************************
     Starting hand 2D plotter component
\***************************************************************
PyTorch version 1.13.1+cu117
Device count 1
/home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/fotis/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Emitting ninja build file /home/fotis/.cache/torch_extensions/py37_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
tcp://192.168.1.193:10005
tcp://192.168.1.193:11005
Trying to make simulation
Not connected to PVD
+++ Using GPU PhysX
PyTorch version 1.13.1+cu117
Device count 1
/home/fotis/Downloads/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/fotis/.cache/torch_extensions/py37_cu117 as PyTorch extensions root...
Emitting ninja build file /home/fotis/.cache/torch_extensions/py37_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
\***************************************************************
     Starting allegro hand sim operator component
\***************************************************************
/home/fotis/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link Base link (index: 0) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/fotis/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_3.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/fotis/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_7.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/fotis/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_11.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))
/home/fotis/miniconda3/envs/openteach_isaac/lib/python3.7/site-packages/ikpy/chain.py:60: UserWarning: Link joint_15.0_tip (index: 5) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive
  warnings.warn("Link {} (index: {}) is of type 'fixed' but set as active in the active_links_mask. In practice, this fixed link doesn't provide any transformation so is as it were inactive".format(link.name, link_index))

\***************************************************************
     Starting calibration component
\***************************************************************

Calibration file already exists. Do you want to create a new one? Press y for Yes else press EnterSimulation created
Loading asset 'allegro_hand_description/urdf/model_only_hand.urdf' from '/home/fotis/PycharmProjects/Open-Teach/openteach/components/environment/assets/urdf/'
[Error] [carb.gym.plugin] Failed to resolve visual mesh '../meshes/table/robot_stand.obj'
Loading Assets
Env suite is not None
\***************************************************************
     Starting Allegro_Sim environment component
\***************************************************************
Start controlling the Simulation hand using the Oculus Headset.

End Effector Position [ 1.         1.2        0.        -0.7071068 -0.7071068  0.
  0.       ]

ligerfotis, 2024-07-19:
My apologies. This was not eventually an error that prevented Isaac gym from starting. Just because the different processes printed in the console in a random order I did not see the `calibration file already exists. Do you want to create a new one? Press y for Yes else press Enter` prompt, which waits for enter to be pressed to start the simulation.

closed 2024-07-19 by ligerfotis

Li-Zizhen, 2024-08-08:
I am also facing the same issue with my Quest 3. In my Isaac Gym space, I can see it tracking my hand gestures in the upper left corner, but the hand in Isaac Gym doesn't move with my hands. It remains stuck with the end position:
```
End Effector Position [ 1.0002534e+00  8.7997133e-01  1.4123693e-04 -7.0756638e-01  7.0663381e-01  4.2378055e-03 -6.9278473e-04] 
```
Could you please let me know how you resolved this problem? I also see the message:
```Calibration file already exists. Do you want to create a new one? Press y for Yes else press Enter```
However, this is immediately followed by other commands, so it doesn't wait for me to press enter. Thank you very much!

Li-Zizhen, 2024-08-08:
Hi, @ligerfotis . I am also facing the same issue with my Quest 3. In my Isaac Gym space, I can see it tracking my hand gestures in the upper left corner, but the hand in Isaac Gym doesn't move with my hands. It remains stuck with the end position:
```
End Effector Position [ 1.0002534e+00  8.7997133e-01  1.4123693e-04 -7.0756638e-01  7.0663381e-01  4.2378055e-03 -6.9278473e-04]
```
Could you please let me know how you resolved this problem? I also see the message:

```Calibration file already exists. Do you want to create a new one? Press y for Yes else press Enter```
However, this is immediately followed by other commands, so it doesn't wait for me to press enter.

ligerfotis, 2024-08-12:
Hi!
TL;DR: Just press enter when all threads are up and running.

There are a bunch of threads getting started and the terminal is filled with messages from everyone. One of them prompts you to what I have written above. That means it just wants you to press enter.
I think you can ignore the messages for now or try to suppress them in the code.