aadhithya14/Open-Teach #21: About collecting data

chiawenwang, 2024-07-25:
Hi！ When I run `the python data_collect.py robot=realman demo_num=1` on my own robot, It stopped with the following outputs:
```

connected to realman arm: left_
Initializing:left_
***************************************************************
     Starting left_arm_cartesian_states component
***************************************************************
Checking if the keypoint port is active...
2024-07-26 05:38:43 - robotic_arm_package.robotic_arm - INFO: Set_Modbus_Mode:0
connected to realman arm: left_
Initializing:left_
***************************************************************
     Starting left_arm_joint_states component
***************************************************************
Checking if the keypoint port is active...
```
some parts of configs is like below:
```
robot_configs: {'_target_': 'openteach.robot.rmins_left.RMINSLeft', 'prefix': 'left_', 'record_type': True}
recorder_function_key: joint_states
storage_path: extracted_data/demonstration_1
Starting to record RGB frames from port: 10006
robot_configs: {'_target_': 'openteach.robot.rmins_left.RMINSLeft', 'prefix': 'left_', 'record_type': True}
recorder_function_key: cartesian_states
storage_path: extracted_data/demonstration_1
```

chiawenwang, 2024-07-26:
It is likely related to the ZMQKeypointSubscriber, specifically in bimanul
```
    def get_cartesian_state_from_socket(self):
        self.cartesian_state_subscriber = ZMQKeypointSubscriber(
                host = '10.19.216.156', 
                port = 8116,
                topic = 'cartesian'
            )
        cartesian_state = self.cartesian_state_subscriber.recv_keypoints()
        cartesian_state_dict= dict(
            cartesian_position = np.array(cartesian_state, dtype=np.float32),
            timestamp = time.time()
        )
        return cartesian_state_dict
```

Why is  the timestamp unnecessary on robot like franka?
```
    def get_cartesian_position(self):
        return self._controller.get_arm_cartesian_coords()
```

chiawenwang, 2024-07-26:
what is the difference among 
`'cartesian_states': self.get_cartesian_state_from_socket,` `'actual_cartesian_states': self.get_robot_actual_cartesian_position` and `'commanded_cartesian_state': self.get_cartesian_commanded_position` ?

aadhithya14, 2024-07-26:
He @chiawenwang , thanks for using openteach. The difference between get_cartesian_from_socket , actual_cartesian_states and commanded_cartesian_state is that , 
get_cartesian_from_socket actually saves the data at around 30 Hz. 
actual_cartesian_states saves the data at the saving frequency mentioned in the data file around 90 Hz.
Commanded cartesian states saves the commanded data. By commanded data I mean it is actually the data which we send from the VR. Actual cartesian states are the states which the robot actually reaches. This just to consider the fact that if the robot stops midway because of an object before reaching the commanded position, that is considered actual position.

aadhithya14, 2024-07-26:
If you are unable to save the robot data, it is good to test the robot wrapper once and print out if the robot positions are received properly. It would be great if you could share the screenshot of how are you publishing the information to the particular socket.

chiawenwang, 2024-08-03:
Thank you for your replying! I just decide to use the only joint states and image to test my model right now. By the way, I also found there would be some problem when using multiple realsense camera together. I fixed it just by change the connecting line and plug in and out. 

It's really a good work!, and in the following training process, I will share the further work detail and problem I meet!

closed 2024-08-03 by chiawenwang