aadhithya14/Open-Teach #22: Is it possible to remove the camera streaming window and just use pass-through?

jmcoholich, 2024-07-30:
I have gotten OpenTeach up and running but I would like to view the robot only through the VR passthrough. The screen for camera streaming is often in the way, however. Is there a way to just disable this screen? 

Best,
Jeremiah

aadhithya14, 2024-07-31:
Yes you can move the screen as soon as you wear the VR by pinching two fingers (index and thumb of right hand) and moving the screen with your head. But to totally remove the screen if you want, you have to alter the unity code.

jmcoholich, 2024-08-12:
Thanks for your response. I am unable to move the screen inside the teleop application with pinching or with the hand remote. Moving the Quest UI around does not affect the screen placement in the teleop app. I will try changing the Unity code.

closed 2024-10-22 by jmcoholich