aadhithya14/Open-Teach #23: Questions on coordinate system-related issues and precision of quest3 hand detection.

dannyXSC, 2024-08-14:
Hello, i am studying your framework recently. I have some confusion regarding the transformation from hand coordinate to robot arm coordinate such as [the code here](https://github.com/aadhithya14/Open-Teach/blob/46150337018d85fc7f3e8fd06e72da834fa65683/openteach/components/operators/kinova.py#L282).

Besides, I evaluate the data obtain from [TransformLeftHandPositionCoords](https://github.com/aadhithya14/Open-Teach/blob/46150337018d85fc7f3e8fd06e72da834fa65683/openteach/components/detector/left_keypoint_transform.py#L10C7-L10C38), and find the accuracy of the data detected by quest3 is very low. Have you ever met the problem?

aliyah-smith, 2024-08-19:
I'm also wondering how this transformation is determined. And how it should be determined for other robot setups? Any insights appreciated.

aadhithya14, 2024-08-22:
Hi, thanks for using Openteach and sorry for my late reply. In Openteach the hand coordinated returned from the VR are in left hand coordinate frame so what we do is we create a local frame with those keypoints and this local frame is also in left hand coordinate frame. Once we have this frame we convert this local frame to Right hand coordinate frame. This matrix is for that reason.

closed 2024-10-05 by aadhithya14