aadhithya14/Open-Teach #27: Is it possible to teleoperate with the remote instead of hand tracking?

jmcoholich, 2024-10-22:
Hi,

We are teleoperating a Franka Panda robot with just a gripper. We have found that the gripper actuation (clicking the thumb and pinky together) is unreliable and causes the hand position to shift in the process. This makes grasping objects very difficult. 

It it possible to teleoperate with the Meta Quest remote and use a button for actuating the gripper? This would give us better position tracking and easier gripper actuation. 

If not, could you recommend any other codebase/stack that does this? 

Thanks,
Jeremiah

dannyXSC, 2024-10-23:
If you want to use remote, it should work. The key of this task is to find the position and rotation information of the remote, for which you need to refer to the documentation.
However, it is possible to control the gripper actuation with thumb and Index finger without exerting a strong influence on the position of the hands. 
If you can provide the tasks you want to accomplish and the situations you cannot tolerate, it will be easier for me to help you.

jmcoholich, 2024-10-24:
I see. In order to implement this change, we would have to change the Unity code and the .apk file right? 

> "However, it is possible to control the gripper actuation with thumb and Index finger without exerting a strong influence on the position of the hands."

- Unfortunately, we found this to be false. Everytime we try to actuate the the gripper with our fingers, the hand position shifts. We tried the thumb and index finger, but the fingers are too close and we get false positives on them touching.

jmcoholich, 2024-12-04:
Hi, would you be able to provide some guidelines on how to edit the Unity package and compile it into a new .apk file?

Awilekong, 2025-04-10:
> Hi, would you be able to provide some guidelines on how to edit the Unity package and compile it into a new .apk file? Is a Windows computer required?你能提供一些关于如何编辑 Unity 包并将其编译成新的.apk 文件的指南吗？需要 Windows 电脑吗？
> 
> Our plan is to edit the Franka-Bot-Unity app to remove the camera streaming window and track the right controller instead of the right hand.我们的计划是编辑 Franka-Bot-Unity 应用程序，以移除相机流窗口并跟踪右侧控制器而不是右手。

> > Hi, would you be able to provide some guidelines on how to edit the Unity package and compile it into a new .apk file? Is a Windows computer required?你能提供一些关于如何编辑 Unity 包并将其编译成新的.apk 文件的指南吗？需要 Windows 电脑吗？
> > Our plan is to edit the Franka-Bot-Unity app to remove the camera streaming window and track the right controller instead of the right hand.我们的计划是编辑 Franka-Bot-Unity 应用程序，以移除相机流窗口并跟踪右侧控制器而不是右手。
> 
> Hi, may I kind inquire have you found the solution? we also want to control the robo arm using VR controller.你找到解决方案了吗？我们也想使用 VR 控制器来控制机械臂。 Thanks!  谢谢！

i have the same need, thanks

David-Kingsman, 2025-05-07:
i have the same need, thank you

jmcoholich, 2025-07-25:
Hi all,

We have modified OpenTeach to allow teleoperation with the Quest 3 controller.

The code and apk file can be found here: [https://github.com/jmcoholich/openteach](https://github.com/jmcoholich/openteach)

Let us know if you have any issues; we haven't tested it on other systems yet.

-Jeremiah

closed 2025-08-31 by jmcoholich