aadhithya14/Open-Teach #29: Use only the video streaming in a custom project

GianniRos, 2024-12-13:
Hi, 
Thank you for the project; it’s very interesting! I would like to use only the streaming part. I have a ZED camera attached to my robot, and I want to teleoperate it using the Quest 3. I’ve been trying to stream over the network using GStreamer, but I could only find solutions for Windows, not Android. That’s how I came across your project, which seems perfect for my needs.

Could you help me out? Should I modify the launch_server file with the camera IDs? And in my project, how can I display the stream? Do I need to add planes and apply a specific script to them?

Thanks a lot 
KR

OhhTuRnz, 2025-01-14:
If you already have a ZED Camera, have you checked OpenTelevision?

https://github.com/OpenTeleVision/TeleVision

GianniRos, 2025-01-15:
thanks so much i really appreciate it, i'll try it

David-Kingsman, 2025-05-06:
Did you solve the problem? Could you share your process?

GianniRos, 2025-05-06:
No, I'm sorry I followed an other path. 
I obtained a result just by streaming the video on the network in jpeg format on two square panels in Unity.
I decreased a lot the resolution because it was an high payload. 
In the end I left the project, because the teleoperation of the robot was not smooth and the quality of the video was orrible.
I still have to finish it