aadhithya14/Open-Teach #34: Using Open-Teach to control single XArm6

Akumar201, 2025-02-02:
Hello, I just found out this repository it looks interesting to me. I want to try controlling my XArm6 using it,  but the issue is I have only one arm, so wanted to know If you can point me towards a config or file which I can edit or use as a template to control the single robotic arm using Open-Teach. Any help would be much appreciated. 

Thank You

aadhithya14, 2025-02-18:
We have already integrated XArm. You can look at the bimanual_right and bimanual_left for controlling right arm with the hand and left arm with left hand.

David-Kingsman, 2025-05-06:
Did you solve the problem? Could you share your process?

lorepieri8, 2025-05-26:
@aadhithya14 What is the suggested way to run a single arm? What files/configs need to be changed?