aadhithya14/Open-Teach #35: How to integrate any other robotic arm for teleoperation

Akumar201, 2025-02-05:
Hello, I would like to know how can I use Open-Teach to control robotic arm that are different from what is available in current library like agileX, Realman. This could be a great question for someone that want to start for using it for multiple data collection out there.  Any guidance would be much appreciated.

aadhithya14, 2025-02-18:
Hi @Akumar201 , thanks for using Openteach. You can integrate any new robot by adding a new robot wrapper as well as a operator wrapper. we have provided sample script to integrate new robot easily

David-Kingsman, 2025-05-06:
> 您好，我想知道如何使用 Open-Teach 来控制与 agileX、Realman 等现有库中提供的不同的机械臂。对于想要尝试使用它进行多数据收集的人来说，这是一个很好的问题。如有任何指导，我将不胜感激。

Did you solve the problem? Could you share your process?