aadhithya14/Open-Teach #36: Failed to resolve visual mesh '../meshes/table/robot_stand.obj'

dddkkkuuu, 2025-02-18:
When I run python teleop.py robot=allegro_sim sim_env=True, I get the error [Error] [carb.gym.plugin] Failed to resolve visual mesh '../meshes/table/robot_stand.obj', and in the VR environment, the robotic hand can only move its fingers.   
I hope to get an answer.

aadhithya14, 2025-02-18:
Hi @dddkkkuuu. Thanks for using Openteach. You are right allegro_sim will only be able to move its fingers. If you want to move the hand from one place to another use robot=moving_allegro_sim.

OrderJade, 2025-02-18:
> When I run python teleop.py robot=allegro_sim sim_env=True, I get the error [Error] [carb.gym.plugin] Failed to resolve visual mesh '../meshes/table/robot_stand.obj', and in the VR environment, the robotic hand can only move its fingers. I hope to get an answer.

you can download the assert source from other repo, https://github.com/Skylark0924/Rofunc/blob/67083ec1e15fab35f67728c18341244d5ddbde61/rofunc/simulator/assets/urdf/table/table.urdf#L44