aadhithya14/Open-Teach #43: Question about bimanual in xarm.

OrigamiOuO, 2025-03-27:
Hello! Thanks for making this great works open-source.

We try to use your repo in controlling a single arm system includes a single xarm and a leap hand. So I have read and changed some of the code and configs. But there are still some questions.

1. The 'pause_status' in '_apply_retargeted_angles()' in 'operators/bimanual_right.py' may be useless. When I pinch my index and ring in VR apk, it will change the color to red but I can still teleop the xarm.
 
2. We start the telop system in this order: 
- In VR start the bimanual.apk and connect the host IP. And the background will be green but not hand pose. 
- Start the teleop.py in host comupter. Now in VR there wil be the hand pose image and the background is green. But we cannot teleop the xarm.
- When I pinch my index the backgound will be red and I can control my xarm, but there will be a violent movement in xarm. It may lead to control error of xarm. But at least I can control xarm if the violent movement do not lead to other problems. I want to know how I can avoid the violent movement.

3. Network: Sometimes when I teleop the xarm there will be some stucks which is that when I move my hand the hand pose in VR and xarm dont follow my hand. I wonder if it is caused by my network environments or the performance of ZMQ. And I suspect it will be the reason for 2. .

lorepieri8, 2025-05-26:
I also see violent Xarm movements, not sure what is causing them as I set velocity and acceleration to low values in the ros_links/bimanual.py.

lorepieri8, 2025-05-27:
The violent movements were due to the mode 1 in the xarm api, commanding servos directly. This mode will use max speed. A wrong default pose in the constants.py file (https://github.com/aadhithya14/Open-Teach/blob/main/openteach/constants.py) can then lead to big fast movements.

Kindred-Yi, 2025-08-26:
For the pause_status issue, I think there’s a bug in get_pause_state_from_hand_keypoints(). Before any pinch, it returns True, but it should be True only after a pinch.