aadhithya14/Open-Teach #47: How to add a new senesor such as the ZED camera instead of the realsense? Could you provide a general procedure how that can be achieved?

David-Kingsman, 2025-05-07:


aadhithya14, 2025-06-02:
Hi @David-Kingsman , u can easily add ZED camera directly into the framework by creating a class for ZEDCamera similar to Realsense camera class in sensors folder. You can take that as the example and construct a similar class for ZEDCamera

aadhithya14, 2025-06-02:
`class ZedMiniCamera(Component):
    def __init__(self, stream_configs, stream_oculus=False):
        np.set_printoptions(suppress=True)
        self._stream_configs = stream_configs
        self._stream_oculus = stream_oculus

        self.rgb_publisher = ZMQCameraPublisher(
            host=stream_configs["host"], port=stream_configs["port"]
        )

        if self._stream_oculus:
            self.rgb_viz_publisher = ZMQCompressedImageTransmitter(
                host=stream_configs["host"],
                port=stream_configs["port"] + VIZ_PORT_OFFSET,  # noqa: F405
            )

        self.depth_publisher = ZMQCameraPublisher(
            host=stream_configs["host"],
            port=stream_configs["port"] + DEPTH_PORT_OFFSET,  # noqa: F405
        )

        self.timer = FrequencyTimer(CAM_FPS)  # noqa: F405
        print("Starting Zed Mini camera")
        self._start_zedmini()

    def _start_zedmini(self):
        # Create a ZED camera object
        self.zed = sl.Camera()

        self.init_params = sl.InitParameters(
            coordinate_units=sl.UNIT.METER,
            coordinate_system=sl.COORDINATE_SYSTEM.RIGHT_HANDED_Y_UP,
        )
        # Set configuration parameters
        self.init_params.camera_resolution = sl.RESOLUTION.HD1080
        self.init_params.camera_fps = 30

        # Set white balance to 4600K
        # self.zed.set_camera_settings(sl.VIDEO_SETTINGS.WHITE_BALANCE, 4600)
        # Reset to auto exposure
        self.zed.set_camera_settings(sl.VIDEO_SETTINGS.EXPOSURE, -1)

        # Open the camera
        err = self.zed.open(self.init_params)
        print("Error", err)
        if err != sl.ERROR_CODE.SUCCESS:
            exit(-1)

    def get_rgb_depth_images(self):
        frames = None
        while frames is None:
            image = sl.Mat()
            if self.zed.grab() == sl.ERROR_CODE.SUCCESS:
                # A new image is available if grab() returns SUCCESS
                self.zed.retrieve_image(image, sl.VIEW.LEFT)  # Retrieve the left image
                frames = np.asanyarray(image.get_data())
                # point_cloud = sl.Mat(res.width, res.height, sl.MAT_TYPE.F32_C4, sl.MEM.CPU)  # noqa: E501
                # depth_frame =  self.zed.retrieve_measure(point_cloud, sl.MEASURE.XYZRGBA,sl.MEM.CPU, res)  # noqa: E501
                timestamp = self.zed.get_timestamp(
                    sl.TIME_REFERENCE.IMAGE
                ).get_seconds()  # Get
                # print("timestamp",timestamp)
        return frames, timestamp

    def stream(self):
        self.notify_component_start("zedmini")
        print("Started the Zed Mini camera:")
        print(
            "Starting stream on {}:{}...\n".format(
                self._stream_configs["host"], self._stream_configs["port"]
            )
        )

        if self._stream_oculus:
            print("Starting oculus stream on port:\n".format())

        while True:
            self.timer.start_loop()
            color_image, timestamp = self.get_rgb_depth_images()
            self.rgb_publisher.pub_rgb_image(color_image, timestamp)
            if self._stream_oculus:
                self.rgb_viz_publisher.send_image(
                    rescale_image(color_image, 2)
                )  # 640 * 360
            self.timer.end_loop()`